SchrodingersGat/KiBoM #6: quoted commas in csv

bootchk, 2016-08-15:
When submitting csv to Digikey, in the Description (or Value?) column I had text having a comma.  In the csv, the whole text was quoted.  Digikey seems to recognize the quoted comma as a column separator.  I don't know enough about the csv specification to know where the issue belongs, but probably with Digikey, and you should dismiss this issue as a hint to users not to use commas in their fields.

Thanks again.

SchrodingersGat, 2016-08-15:
Not sure how to deal with this one, if Digikey isn't treating the commas properly then I'd suggest just not putting an extra comma there ;)

bootchk, 2016-08-15:
Yes, not your problem.

 I figured you knew the code much better than I did and could fix the other issue quickly.

Again, thanks so much.

If you haven’t seen it, I also use MadParts (on github) with KiCad.

closed 2016-08-26 by SchrodingersGat