SchrodingersGat/KiBoM #11: TypeError: a bytes-like object is required, not 'str'

pointhi, 2016-11-15:
```
Errormessage:
Traceback (most recent call last):
  File "/home/thomas/Projekte/KiBoM/KiBOM_CLI.py", line 103, in <module>
    pref.Write(ini)
  File "/home/thomas/Projekte/KiBoM/KiBOM/preferences.py", line 205, in Write
    cf.write(configfile)
  File "/usr/lib/python3.5/configparser.py", line 916, in write
    self._sections[section].items(), d)
  File "/usr/lib/python3.5/configparser.py", line 920, in _write_section
    fp.write("[{}]\n".format(section_name))
TypeError: a bytes-like object is required, not 'str'
```

pointhi, 2016-11-15:
Ok, found error. You need to run it with python2.7

closed 2016-11-15 by pointhi

SchrodingersGat, 2016-11-15:
Can it be generalised so it works with 3.x also?

pointhi, 2016-11-15:
Likely. You need to rewrite the project so it's both python2.7 and python3 compatible. For that you have to identify constructs which only works in one of both versions, and rewrite them either generic (preferred) or with an condition which executes the correct code path based on the python version.

Like I have done for [kicad-footprint-generator](https://github.com/pointhi/kicad-footprint-generator) which officially supports (2.7, 3.2, 3.3, 3.4, 3.5) ([see here](https://travis-ci.org/pointhi/kicad-footprint-generator))