SchrodingersGat/KiBoM #13: Group References into ranges

justyn, 2017-04-27:
For boards where there are many identical components, it would be good to have the option to group references together like KiField does.

For example instead of:

R1, R4, R5, R6, R7, R8, R9, R10

you could have:

R1, R4-R10

SchrodingersGat, 2017-05-03:
I like that - it would be nice to have that as an option flag (I wouldn't want to use such a feature all the time)

schneidersoft, 2017-05-31:
Let me know if you'd rather want R8,R11,R12,R21-R42 or R8,R11-R12,R21-R42
... should groups of 2+ get a - or only groups of 3+

SchrodingersGat, 2017-05-31:
I think 3+ makes sense. Are you going to implement this? :)

schneidersoft, 2017-05-31:
> 5493e4bb","name":"GitHub"},"entity":{"external_key":"github/Schroding
> ersGat/KiBoM","title":"SchrodingersGat/KiBoM","subtitle":"GitHub
> repository","main_image_url":"https://cloud.githubusercontent.com/ass
> ets/143418/17495839/a5054eac-5d88-11e6-95fc-
> 7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent
> .com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-
> b52498112777.png","action":{"name":"Open in
> GitHub","url":"https://github.com/SchrodingersGat/KiBoM"}},"updates":
> {"snippets":[{"icon":"PERSON","message":"@SchrodingersGat in #13: I
> :)"}],"action":{"name":"View
> Issue","url":"https://github.com/SchrodingersGat/KiBoM/issues/13#issu
> ecomment-305339300"}}}

Yup. Should be up momentarily.

KarlZeilhofer, 2017-08-03:
@schneidersoft how about the implementation of this feature?

I've one additional feature request: 
```
C201 C204-C211 C308 C310 C313 C412 C601 C602-C610 C613
```
would be still a very wide cell in the spread sheet. I would prefer a line break after N entries, e.g. 5
```
C201 C204-C211 C308 C310
C313 C412 C601 C602-C610
C613
```

Once I've written my own BOM-tool in C, which had this feature, it was exported with \n, which 
was displayed correctly in libre office calc.

schneidersoft, 2017-08-03:
Ok. so you want to word wrap to N words. where each entry C201, C202 etc. is a word.
your example works when N is 4.
What happens when N is 2?

C201 C204-
C211 C308
C310 C313
C412 C601
C602-C610
C613

When N is 3
C201 C204-C211
C308 C310 C313
C412 C601 C602-
C610 C613

Uggly much?
Are you suggesting we alter the hyphenation?
N=2:
C201 C204
C205-C211
C308 C310
C313 C412
C601 C602
C603-C610
C613

KarlZeilhofer, 2017-08-03:
I would prefer the last option. setting N to 2 is not used in daily life, I think. 

If we use N=5, then in worst case there are some lines with only 4 entries...

```
C201 C204-C211 C301 C308 
C310 C313 C412 C601 
C602-C610 C613 C617 C619
C621
```

Are you going to implement this?

schneidersoft, 2017-08-03:
Yes. I have implemented it.

KarlZeilhofer, 2018-03-28:
@schneidersoft Thanks! Where can I find it and how can I use it?

KarlZeilhofer, 2018-03-28:
@SchrodingersGat I saw, this ferature wasn't merged into this master branch, why not?

schneidersoft, 2018-03-28:
you can always pull from my repo. But it would be better to get the changes merged into this one.

SchrodingersGat, 2018-03-28:
@schneidersoft I can't see an open PR for this feature - have I mistakenly closed it, or have you not opened one?

schneidersoft, 2018-03-29:
PR #19 incorporates this feature and a few others.
I made a bunch of changes one after the other., and only one PR. I don't know if that's the correct way of doing things.

SchrodingersGat, 2018-04-05:
#19 is merged now - OK to close this issue?

schneidersoft, 2018-04-05:
Sure

closed 2018-04-05 by SchrodingersGat

KarlZeilhofer, 2018-05-27:
Looks great now. Here's an example:
[image]

3 things I'd like to notice:
1. This feature is not documented in the README.md
2. `alt_wrap` only works, if `use_alt` is set to 1. For documentation only, this is quite okay, but in some cases, where we do a manual pick and place, I like to check every reference with a pencil on paper. But this doesn't work any more, if the references are grouped in ranges. 
3. perhaps the options should have a better naming: 
   `wrap_ref_count` instead of `alt_wrap` and  
   `use_ref_ranges` instead of `use_alt`