SchrodingersGat/KiBoM #20: Explicitly output fields as text

schneidersoft, 2017-05-31:
Importing csv/tsv files in libreoffice correctly requires that you set the type of each column manually.
Otherwise texts like the footprint "0805" will be imported as a number and turned into 805.

It would be helpful to have Kibom wrap fields in quotes to allow import into libreoffice with the "import quoted fields as text" feature.

Naturaly those field that do contain numeric data like the Qty and Build Quantity should not get quotes.

suzizecat, 2017-07-04:
I've look into it, and if i understand correctly, the part of the code that prepare the fields to be written is in `component.py` and in the function

    def getRow(self, columns):
        row = []
        for key in columns:
            val = self.getField(key)
            
            if val is None:
                val = ""
            else:
                val = u'' + val
                val = val.encode('utf-8')
                
            row.append(val)
            
        return row

that use this one :
        
    def getField(self, field):
    
        if not field in self.fields.keys(): return ""
        if not self.fields[field]: return ""
        return u''.join((self.fields[field]))

which is line 311. 

But, as i'm not a Python specialist, i don't really understand the last line and i'm not even sure that this is the right place. 

Could someone tell me if i'm right ? 
I may take some time to add this feature later. BTW, should it use an option in the command line or use the option system provided by the ini file (which i don't know how to use btw)

schneidersoft, 2017-07-04:
The last line just says make an empty unicode string and add the
contents of self.fields[field] to it.

We should probably be explicitly specifying an encoding too, so using 
.encode('utf-8') when we output the unicode string.

The problem really isn't with python thought, its that we want the
columns in the csv file to be wrapped in quotes. This will cause
calc/excel to think twice about converting order numbers that look like
00012345 to numbers like 12345...

idealy we should be able to open a kibom output csv in calc, export it
again, and have two copies of the exact same file.

suzizecat, 2017-07-05:
Sure that's what i want to do, but i need to understand the Python code properly to find the right place to add the quotes. BTW you only have to add one apostrophe (`'`) at the beginning of the field to make it a text field.

schneidersoft, 2017-08-29:
I have noticed that libreoffice allows you to select 'import quoted fields as text'
This sounds like what we need.
Libre office exports spreadsheets as tsv like this:
1	"Wuerth"	"744732152"	"Wuerth"	"744732152"

whereas kibom produces output like this:
R20	1	51k	0805	YAGEO	RC0805FR-0751KL

So really I thin we should be looking at the csv python package and how to teach it to use quotes.
probably something to do with line 37 csv_writer.py
writer = csv.writer(f, delimiter=delimiter, lineterminator="\n")

gonna go read the docs.

suzizecat, 2017-08-29:
Yep, it's in there. I you look at it, you'll find the spot where the fields are written. Here you cant simply add commas before/after the value without problem. Just make sure to add this feature as an option 😉

schneidersoft, 2017-08-29:
turns out you can use:
writer = csv.writer(f, delimiter=delimiter, lineterminator="\n", quoting=csv.QUOTE_NONNUMERIC)

To force quotes.
problem is that it quotes everything even the quantity field. I think this is because getRow encodes to utf8.