SchrodingersGat/KiBoM #26: Revision/configuration vs variant

diegoherranz, 2018-01-13:
Hi,

First of all, thanks for this tool. I haven't used it yet, but looks perfect for my needs.

While reading the documentation, I've seen the `--revision` parameter and I find it misleading. Revision to me, means version, as in v3, v4.1, rev2 or something like that. Also configuration is used in some places for this, which is a bit better but I think the industry standard for this is _variant_ which in fact is the word you use in the help to explain it.

If you wanted to use the word variant instead, I can provide a PR. At least for the docs, don't know if you would want to change the API too (or anything at all).

Thanks!

SchrodingersGat, 2018-01-13:
You're right, variant is a superior term. Feel free to submit a PR for this :)

diegoherranz, 2018-01-15:
I was planning to call the new option `--variant` instead of `--revision`. However, what about the short option? `-v` is used for `--verbose`, so a few options:
a) keep `-r` for `--variant`
b) use `-v` for `--variant` and `-V` for `--verbose`
c) use a different one for `--variant`. Maybe `-f` from _flavour_? Other?

What do you think? I think I prefer a), and then b).

SchrodingersGat, 2018-01-16:
a) I think. `-v` is pretty much always verbose

PR #28 (2018-01-28): Use "variant" instead of "revision"
Fixes #26
closed 2018-01-28 by SchrodingersGat