SchrodingersGat/KiBoM #36: References starting with + are DNF?

KarlZeilhofer, 2018-07-26:
I want to export a BOM, where an internal part number is a primary key (unique) in the exported list. 

But test points, which all have the same part number (`NAP`), are grouped into 2 different rows:

```
36 |   | Montageloch | t14_mech | MH301 P1601 S301 TP305-TP306TP701-TP702 TP801 TP901 TP1001TP1101 TP1201 TP1301 TP1401 TP1501TP1601 | Montageloch | MH_DK_M3_4mm-DSD-special | t14_mech | 16 |   | NAP
78 |   | Testpunkt | t14_mech | TP301-TP304 TP307-TP308 | +5V | LSP13 | dsd-cc-eagle | 6 (DNF) |   | NAP

```

Can this "feature" be turned off?

trentks, 2019-07-05:
Hi @KarlZeilhofer,

**Regarding your title:**
References in the `fit_field`, that are starting with '+' will, by design **only** be included in variants which they're explicitly named. Could you clarify the issue? Is your proposition that there be an over-ride where a complete list is printed regardless of variant? If this is the case, then you might find the `ignore_dnf` field of the bom.ini configuration worthwhile looking at as it provides this functionality.

**Regarding your content:**
I don't see how this is related to the title, and it's hard to see your groupings going on here. Could you provide some headings, or a full BoM export file that better exemplifies your conflict?

Cheers.

KarlZeilhofer, 2019-07-07:
Hi, 

I'm not doing here anything with `fit_field`. 

Our symbols in the schematic get a custom field named `InternalName`. I'd like to export a BOM, where all the rows are grouped by this field. In the mentioned case, both lines with the `InternalName = NAP` (not a part) should be grouped into a single line. 

But it seems, that symbols with it's field value starting with a `+` are recognized as a variant and therefore classified as `DNF`. One of our test points has the value `+5V`. But it should be grouped into the `NAP`-line. 

I found this in the documentation:
> Note: If the Value field for the component contains any of these values, the component will also not be included

This belongs to the various names for "do not fit". But it seems, that also the `+`-sign is also evaluated, even though it is not in the `fit_field`. It is in `value = +5V`. 

It seems, there is some inconsistency of the implementation with the documentation. `+` and `-` should be evaluated only in the `fit_field`, but it also happens to be in the `value` field.

KarlZeilhofer, 2019-07-07:
PS: I updated the title. It was misleading and wrong.

KarlZeilhofer, 2019-07-07:
PPS: I updated the table of the original post and added headings to the columns

KarlZeilhofer, 2019-07-07:
P3S: It seems, that problem is on my side. I had this in the ini-file:
```
; If 'ignore_dnf' option is set to 1, rows that are not to be fitted on the PCB will not be written to the BoM file
ignore_dnf = 0

; Field name used to determine if a particular part is to be fitted
fit_field = Value
```
I didn't know/remember, that that field was configurable. 

I think we can close this now.

closed 2019-07-07 by KarlZeilhofer