SchrodingersGat/KiBoM #41: Ignore virtual components

leoheck, 2018-09-05:
Virtual footprints should be ignored by default. I suggest a flag (`-a`) to show all.
For example, I have a footprint that is a logo. It should not be on BOM because I don't need to by a logo :)

easyw, 2018-09-05:
virtual footprints are often used to show/hide mechanical parts (i.e. enclosure, standoff etc.) and they need to be in BOM list.

SchrodingersGat, 2018-09-05:
@leoheck you can add a filter to the bom settings file to remove any items you do not want to appear in the bom output. This is pretty flexible and should allow you to achieve what you want. As @easyw says there are certainly good reasons for including parts that don't actually have a footprint associated.

leoheck, 2018-09-05:
@SchrodingersGat thanks for pointing me that..

leoheck, 2018-09-05:
I am using `[REGEX_EXCLUDE]` and it works fine, thanks.

But why it is tab separated? 

It took me a while to make it work and I fixed it even before I could find this comment:
`; Format is: "ColumName	Regex" (tab-separated)`

The best thing to do is to use whitespace separated (spaces, tabs)*

SchrodingersGat, 2018-09-26:
Whitespace does make sense. Submit a PR and I'll merge the heck out of it :)

KarlZeilhofer, 2019-07-07:
> But why it is tab separated?

Perhaps because a regex can contain space characters, but not tabulators?

leoheck, 2019-07-07:
What? I didn't get it.

leoheck, 2019-07-07:
@KarlZeilhofer I see what you said now. I think it is not a big issue.

SchrodingersGat, 2019-07-09:
Closed by https://github.com/SchrodingersGat/KiBoM/pull/61

closed 2019-07-09 by SchrodingersGat