SchrodingersGat/KiBoM #43: Does not add part without footprint

leoheck, 2018-09-23:
I have 4 parts without a footprint, but with a field, I called MPN (manufacturer part number).
How can I include them on the BOM?

SchrodingersGat, 2018-09-23:
Are they not already included? I did not think that the footprint field is necessary for the item to be included in the BOM?

leoheck, 2018-09-24:
No, they were not included. I also tried to put on [REGEX_INCLUDE]. 
Do you think it can be related to their reference designator?
The shield is using the J3 RefDes and the headers are J3M1, J3M2, J3M3, J3M4.

SchrodingersGat, 2018-09-26:
Can you send an example schematic so I can test?

trentks, 2019-07-07:
Hi @leoheck,

While you're on the job looking at your #41 with #61, I can't replicate this either. I do design work for electrical cabinets in eeschema, where no parts have PCB footprints, and KiBoM creates the BoMs just fine. Can you try the newest master, remove your bom.ini (start fresh), and see if you can replicate this? Or if you can get me a schematic with the issue, I can investigate further for you.

Cheers.

leoheck, 2019-07-08:
Sure, I will look into this tomorrow. @trentks

trentks, 2019-08-12:
Any update there @leoheck?

SchrodingersGat, 2019-09-19:
Hi @leoheck please re-open this issue if it is still not working for you

closed 2019-09-19 by SchrodingersGat