SchrodingersGat/KiBoM #49: Force part insertion in BOM

microcris, 2019-02-26:
Hi there!

I have several parts that I want to include in my BOM but those parts contains words in their footprints or in their library names that makes them excluded.
Example:
Test points:
my_library:TP_SMD_5019
Connectors:
my_library:DSUB-9_Female_Horizontal_P2.74x2.54mm_EdgePinOffset10.07mm_Housed_MountingHolesOffset11mm
Connector_Phoenix_MC:PhoenixContact_MC_1,5_3-GF-3.5_1x03_P3.50mm_Horizontal_ThreadedFlange_MountHole

Anyting that contains the words "test point" or "mount hole" is excluded and it seems that there is no option to bypass that.

In my case, I use a field caled "Part Number" with the device part number.
Is there any chance to force any device that contains a value in the field "Part Number" to be included in the BOM?

Best regards,
Cristiano Rodrigues

SchrodingersGat, 2019-03-01:
Hi,

Try looking in the `bom.ini` file which is created in the project folder.

Under the heading ```[REGEX_EXCLUDE]``` you should be able to remove the lines which are excluding your parts.

SchrodingersGat, 2019-09-19:
Hi @microcris I'll close this issue, please re-open if you feel it is not addressed

closed 2019-09-19 by SchrodingersGat