SchrodingersGat/KiBoM #52: Multiple Config values for same part

cpattersonic2, 2019-04-18:
My schematic is organized by manufacturer (manf) and manufacturer part number (panf_pn) fields. Also we are using the Config field for multiple configurations (-no_batt, -no_12V, -no_5V)
Occasionally we have a part (example resistor Manf = Yageo, Manf_PN= 08051k) that is used in multiple places and has multiple configs. Some update in the last year now prevents parts with 'conflicted_field' to no longer be printed to the BOM. Anyway to fix this?
Thanks!

trentks, 2019-07-01:
Hi @cpattersonic2,

Have you some example project I can look at? Does this problem still persist with the latest master?

Cheers.

trentks, 2019-07-04:
I've just realised that these configuration variants that @cpattersonic2 mentions are all prefixed with '-', so this is likely the same issue as #53 and #55. Something that was fixed by #37, but for which a documentation issue persists but is fixed with #60.

When #60 is merged, all of #52, #53, and #55 should be closable.

SchrodingersGat, 2019-07-04:
Closed in https://github.com/SchrodingersGat/KiBoM/pull/60

closed 2019-07-04 by SchrodingersGat