SchrodingersGat/KiBoM #53: Variants with "-" prefix are DNF regardless of variant specified

adamclaassen, 2019-04-18:
There seems to be a bug in the "exclude-from-variant" feature. Any part with the `-` prefix in the `fit_field` is ignored, regardless of the variant configuration passed. 

Here's an example of the issue:

Ref | fit_field
----- | -------
C1 | 
C2 | -foo
C3| +foo

With no variant flag, only C1 appears. With the option `-r foo` passed, C1 and C3 appear. 
From the readme, I'd expect C1 and C2 to appear with no flag passed, right?

I'm running Manjaro Linux with KiCAD 5.1.0, Python 2.7.16, and the latest version of KiBOM as of writing (e4df1e8). I've also attached the example project I used to test this. [test-project.zip](https://github.com/SchrodingersGat/KiBoM/files/3096495/test-project.zip)

trentks, 2019-07-01:
Hi @adamclaassen (and @EdwinSD from #55)

I have tested this, and failed to replicate using python2 or python3 (albeit I am using 4d221d5e9372be7d1b89ad8c8bcb48c8332e62ec which is currently ahead of master here until @SchrodingersGat merges #60). This shouldn't be any different in this matter to master here though (16fda208b9104e9ce61c0b46ca7bed3a67669b3e). The "-" functionality works as described (including the component unless the relevant variant causes it to be excluded).

Please note that there was a rather major merge some 2 days ago when @AngusP's #37 pull request was merged which altered a great deal of things that would have been related to this, and may have inadvertently fixed this issue. Could you guys please confirm that you can still produce this issue against either master here, or master on my fork?

Cheers.

AngusP, 2019-07-01:
@trenkts it would probably also be good to add some kind of regression test for the variant config (and also to test my pull #37 works as intended!) which I’ll get around to adding eventually, unless someone beats me to it, which would prevent this kind of issue

SchrodingersGat, 2019-07-04:
Closed in https://github.com/SchrodingersGat/KiBoM/pull/60

@adamclaassen if you have a chance can you please verify that your requirements are met?

closed 2019-07-04 by SchrodingersGat

adamclaassen, 2019-07-04:
@SchrodingersGat Sorry about the delay - I've tested using f0da5a1 and KiCad 5.1.2 w/ Python 3, everything is working great, thanks!