SchrodingersGat/KiBoM #54: Loss of descriptions in the BOM when a component name includes a '/'

Daniellandry62, 2019-05-09:
In custom atomic parts libraries, some components have a value that includes the '/' character. In this case, KiBoM has difficulty in finding the description of many components included in the same files (.lib and .dcm), even if the component value do not have a '/'.

Faulty librairies are easily find using grep:
$ grep 'F1 "' *.lib | grep '/'
$ linear_voltage_regulator.lib:F1 "SPX1117M3-L-3-3/TR" -600 -375 50 H I L TNN
$ linear_voltage_regulator.lib:F1 "SPX1117M3-L/TR" -605 -505 50 H I L TNN
$ thermistor_ptc.lib:F1 "MF-MSMF110/16-2" -750 -475 50 H I L TNN

This limitation was not in the KiBoM master branch in October 2018... or it can be relative to software upgrade on my side. In october, I was running on MX Linux 17.1 with KiCad 4.0.7 and Python 2.7

Presently, I'm running MX Linux 18.2 with KiCAD 5.0.1, Python 3.7.0, and the latest version of KiBOM as of writing (e4df1e8)

trentks, 2019-06-30:
Hi @Daniellandry62,

I've just fixed a similar string issue in #59 (issue #45) that was caused by the difference between python2 and python3. Could you please check this issue exists against KiCAD latest (5.1.2) calling it with python3, and python2? It is likely that this difference has caused your problem, rather than any change to this software since e4df1e87dc6d16b85d934919a493a7ca3627eb08.

Or alternatively if you could provide me with a problematic .lib/.dcm I can check and debug myself. I do not know how much (if any) testing has been done against python3, so it would be good to iron out these bugs.

Cheers.

Daniellandry62, 2019-07-02:
Hi @trentks ,

I did some tests with python2 and python3 with KiCAD 5.0.1 (latest version available on MX Linux 18.2) and the issue is now solved. I provide you an original problematic library (.lib /.dcm) in case you want to do additional tests. For information, custom atomic parts libraries was done using kicadLibCreator.

I really like KiBoM and I appreciate your work. Thank you

[ubios_linear_voltage_regulator.zip](https://github.com/SchrodingersGat/KiBoM/files/3352461/ubios_linear_voltage_regulator.zip)

SchrodingersGat, 2019-07-03:
Thanks @trentks for input and @Daniellandry62 for testing. Closing this issue! Cheers all.

closed 2019-07-03 by SchrodingersGat