SchrodingersGat/KiBoM #55: Any string in "not_fitted" marks components as DNF

EdwinSD, 2019-05-14:
Any string entered in the "not_fitted" field is marked as (DNF)

Script string: 
`python "C:\Program Files\KiCad\bin\plugins\KiBoM-master\KiBOM_CLI.py" "%I" "%O_BOM.html"`

bom.ini:
```
ignore_dnf = 0
fit_field = Fitted
board_variant = default
```
Output generated:

Nr. | References | Value | Quantity Per PCB | Fitted
--|-------------|--------|--------------------|---
63 | U5 | 24LC256 | 1 (DNF) | +CFG1
64 | U14 | AMS1117-3.3 | 1 (DNF) | -CFG1
65 | U11 | CRE1S0505SC | 1 (DNF) | DNF
66 | U18 | FT230XS | 1 (DNF) | RandomText

KiCAD: (5.1.2)-1, release build, on Windows 10 64-bit

trentks, 2019-06-30:
Hi @EdwinSD,

I think this may just be a misunderstanding of intended and expected behavior, maybe a documentation change is required. I will try to explain.

You have flagged `ignore_dnf = 0`. When to use any find of `fit_field` sensibly you need to set `ignore_dnf = 1`. This is because the `fit_field` appropriately determines which items to set as DNF, these are then ignored and the items are included (quantity set to 1 in your example), but are appended with DNF because if you'd set `ignore_dnf = 1` they would have been DNF and not appeared on the BoM.

When you run KiBOM without any variant specified, only the `fit_field`s which are blank will be output to the BoM as these are the 'base' items used across all variants.

--------------
In your given example you will see that if you ran `python "C:\Program Files\KiCad\bin\plugins\KiBoM-master\KiBOM_CLI.py" "%I" "%O_BOM.html" -r CFG1` with `ignore_dnf = 0` then you would be presented with:

Nr. | References | Value | Quantity Per PCB | Fitted
-- | -- | -- | -- | --
63 | U5 | 24LC256 | 1 | +CFG1
64 | U14 | AMS1117-3.3 | 1 (DNF) | -CFG1
65 | U11 | CRE1S0505SC | 1 (DNF) | DNF
66 | U18 | FT230XS | 1 (DNF) | RandomText

If you'd run `python "C:\Program Files\KiCad\bin\plugins\KiBoM-master\KiBOM_CLI.py" "%I" "%O_BOM.html"` with `ignore_dnf = 1` then you would have gotten:

Nr. | References | Value | Quantity Per PCB | Fitted
-- | -- | -- | -- | --

Finally, if you'd run `python "C:\Program Files\KiCad\bin\plugins\KiBoM-master\KiBOM_CLI.py" "%I" "%O_BOM.html" -r CFG1` with `ignore_dnf = 1` then you would get:

Nr. | References | Value | Quantity Per PCB | Fitted
-- | -- | -- | -- | --
63 | U5 | 24LC256 | 1 | +CFG1

-------------

After having read the `PCB Variants` and `Configuration` sections of the `README.md`, do you think there are any changes that could be made to make this functionality more clear?

Cheers.

EdwinSD, 2019-07-01:
Hi @trentks 

Thanks for looking into the issue. I understand what your saying and I've checked the outputs, and they are exactly like you said. 

My expected behavior probably differs from the intended behavior of the script. I would expected a component marked -CFG1, *only* to be excluded when ran with `-r CFG1 `with` ignore_dnf = 1`

For example: Following the example explained in the readme.md

- C1 -> "-production,+test"
- C2 -> "+production,+test"
- R1 -> ""
- R2 -> "-test"

---
Running the script with `-r production`

\# |  References | Fitted
-- |  -- | --
1 |   C2 | +production,+test
2 |  R1 |  

**R2 doesn't show up, but it should**

---
Running the script with `-r test`

\# |  References | Fitted
-- | -- | -- | 
1 |  C2 | +production,+test
2 |  C1 |  -production,+test
3 |  R1 | 

---

Running the script without any variant specified returns:

\# |  References | Fitted
-- | -- | -- | 
3 |  R1 | 

I've commented out the following line
```
; Default PCB variant if none given on CLI with -r
; board_variant = Default
```

---

As you can see, the script doesn't output what is expected, at least from what I understand of what it should do

Thanks again. Regards,

Edwin

trentks, 2019-07-01:
Ah, 

I think I understand @EdwinSD, I'll have a look at this at some stage. I think basically to sum up, variants which are tagged "-" are being excluded when they should not be.

To go back to your original example, I think that "RandomText" should probably be interpreted the same as "+RandomText" though nowhere does it say explicitly that such a thing is supported or what the behavior should be. I will look into this, and implement it this way if it is not already so, and add a note in the README to that effect. But I believe that your "DNF" is behaving as it should, as is your "+CFG1", but perhaps not your "-CFG1" as discussed.

Cheers.

EdwinSD, 2019-07-01:
I've just updated the KiBom to the latest version, deleted bom.ini, so it would be generated again, and it seems that the problem is resolved, except for running it without arguments.

The readme.md states: 
> If the script is run with the flag --variant production then C2, R1 and R2 will be loaded.
> If the script is run without the --variant production flag, then C1, R1 and R2 will be loaded.
> If the script is run with the flag --variant test, then C1, C2 and R1 will be loaded.
> If the script is run with the flags --variant production,test, then C2 and R1 will be loaded.

If I run the script without a variant, only R1 and R2 are loaded. The rest is generated correctly

Regards,

Edwin

trentks, 2019-07-01:
HI @EdwinSD 

That's the expected behavior, no? Not explicitly stated, but expected?

```

- C1 -> "-production,+test"
- C2 -> "+production,+test"
- R1 -> ""
- R2 -> "-test"
```
C1 will be included in test, unless it's a "test,production" variant, in which case it will be excluded. All other cases it is excluded.
C2 will be included in production, and included in test, but is otherwise excluded.
R1 will be included always.
R2 will be excluded only in the case of test.

Running without a variant therefore should include R1, and R2 but exclude the rest. What is your expected behavior?

EdwinSD, 2019-07-01:
Hi @trentks,

I think that's correct, fixing the documentation should resolve this. 

Thanks for your assistance. Keep up the good work! 

Regards,

Edwin

SchrodingersGat, 2019-07-04:
Closed in https://github.com/SchrodingersGat/KiBoM/pull/60

@EdwinSD  can you please confirm that you're happy with this fix?

closed 2019-07-04 by SchrodingersGat

EdwinSD, 2019-07-04:
Hi @SchrodingersGat,

Yes, resolved. I've tried various of my projects and now it's working correctly.

Thanks!