SchrodingersGat/KiBoM #62: Error with KiCAD 5.1.2 [Linux\

Brian245, 2019-07-10:
I've received the error below with KiBOM on several installations.  If I go in and comment out line 207 in the related module and then add missing items to the .ini file that is generated, I can ultimately get KiBOM to work.  
This error below resulted from running KiBOM within KiCAD with the following command line specified:

python "/home/brian/Documents/KiCAD/KiBoM-master/KiBOM_CLI.py" "%I" "%O.html"

Has anyone else had this problem?

Thanks, Brian

Result is below:
------ include -------
Run command:
python "/home/brian/Documents/KiCAD/KiBoM-master/KiBOM_CLI.py" "/home/brian/Documents/KiCAD/Micro_Regulator_1/MicroRegulator1/MicroRegulator1.xml" "/home/brian/Documents/KiCAD/Micro_Regulator_1/MicroRegulator1/MicroRegulator1.html"

Command error. Return code 1
Info messages:
PCB variant:  default

Error messages:
Traceback (most recent call last):
  File "/home/brian/Documents/KiCAD/KiBoM-master/KiBOM_CLI.py", line 202, in <module>
    result = writeVariant(variant, args)
  File "/home/brian/Documents/KiCAD/KiBoM-master/KiBOM_CLI.py", line 76, in writeVariant
    pref.Write(config_file)
  File "/home/brian/Documents/KiCAD/KiBoM-master/bomlib/preferences.py", line 207, in Write
    self.addOption(cf, self.OPT_INCLUDE_VERSION, self.includeVersionNumber, comment="If '{opt}' option is set to 1, the schematic version number will be appended to the filename.")
AttributeError: BomPref instance has no attribute 'OPT_INCLUDE_VERSION'

trentks, 2019-07-10:
Hi @Brian245,

In pull #60, the 'OPT_INCLUDE_VERSION' name was removed in favor of a much more versatile file naming structure.

I would suggest that you pull the latest master, and remove your existing bom.ini file (so that it regenerates).

Let me know how you get on. Cheers.

Brian245, 2019-07-10:
Hi @trenks,
I believe I am pulling the latest master - thank you so much for your swift reply!

trentks, 2019-07-10:
In that case, all you should need to do is remove your existing bom.ini file from your kicad project folder/eeschema schematic folder.

Brian245, 2019-07-10:
The bom.ini file does not get generated unless I hack away at the scripts.

trentks, 2019-07-10:
Ok, I'll have a look at that.

In the mean time then, manually create a bom.ini with the default as per README.md (copied below):

```
[BOM_OPTIONS]
; General BoM options here
; If 'ignore_dnf' option is set to 1, rows that are not to be fitted on the PCB will not be written to the BoM file
ignore_dnf = 1
; If 'number_rows' option is set to 1, each row in the BoM will be prepended with an incrementing row number
number_rows = 1
; If 'group_connectors' option is set to 1, connectors with the same footprints will be grouped together, independent of the name of the connector
group_connectors = 1
; If 'test_regex' option is set to 1, each component group will be tested against a number of regular-expressions (specified, per column, below). If any matches are found, the row is ignored in the output file
test_regex = 1
; If 'merge_blank_fields' option is set to 1, component groups with blank fields will be merged into the most compatible group, where possible
merge_blank_fields = 1
; Specify output file name format, %O is the defined output name, %v is the version, %V is the variant name which will be ammended according to 'variant_file_name_format'.
output_file_name = %O_bom_%v%V
; Specify the variant file name format, this is a unique field as the variant is not always used/specified. When it is unused you will want to strip all of this.
variant_file_name_format = _(%V)
; Field name used to determine if a particular part is to be fitted
fit_field = Config

[IGNORE_COLUMNS]
; Any column heading that appears here will be excluded from the Generated BoM
; Titles are case-insensitive
Part Lib
Footprint Lib

[GROUP_FIELDS]
; List of fields used for sorting individual components into groups
; Components which match (comparing *all* fields) will be grouped together
; Field names are case-insensitive
Part
Part Lib
Value
Footprint
Footprint Lib

[COMPONENT_ALIASES]
; A series of values which are considered to be equivalent for the part name
; Each line represents a tab-separated list of equivalent component name values
; e.g. 'c c_small cap' will ensure the equivalent capacitor symbols can be grouped together
; Aliases are case-insensitive
c	c_small	cap	capacitor
r	r_small	res	resistor
sw	switch
l	l_small	inductor
zener	zenersmall
d	diode	d_small

[REGEX_INCLUDE]
; A series of regular expressions used to include parts in the BoM
; If there are any regex defined here, only components that match against ANY of them will be included in the BOM
; Column names are case-insensitive
; Format is: "ColumName	Regex" (tab-separated)

[REGEX_EXCLUDE]
; A series of regular expressions used to exclude parts from the BoM
; If a component matches ANY of these, it will be excluded from the BoM
; Column names are case-insensitive
; Format is: "ColumName	Regex" (tab-separated)
References	^TP[0-9]*
References	^FID
Part	mount.*hole
Part	solder.*bridge
Part	test.*point
Footprint	test.*point
Footprint	mount.*hole
Footprint	fiducial
```

Brian245, 2019-07-10:
Many thanks @trentks .  I checked the README.md and didn't notice the instructions you mention related to bom.ini  Everything works perfectly when I copy the above into 'bom.ini'
Thanks again!!

trentks, 2019-07-10:
No worries @Brian245,

It's my fault, I'd missed the default bom.ini generation parts of the code in pull #60. I'm writing a patch now to address this issue and correctly generate a default bom.ini that's functional, thanks for reporting it.

Brian245, 2019-07-10:
Thank you again for your help, @trentks!!

trentks, 2019-07-10:
This issue should be solved when #63 is merged. Thanks @Brian245.

trentks, 2019-07-10:
Hi @Brian245,

@SchrodingersGat has merged the fix (#63), I've tested it as per my understanding of the problem---but perhaps you can pull the latest master, confirm, and close out this issue.

Cheers.

Brian245, 2019-07-10:
Hi @trentks,
Just pulled the master, deleted the previous 'bom.ini', and confirmed proper operation and generation of initial 'bom.ini' file.  Many thanks for resolving this issue so swiftly.  All working now!
Cheers!

closed 2019-07-10 by Brian245