SchrodingersGat/KiBoM #67: bom.ini file does not support "True"

peter-jensen, 2019-09-27:
The generated "ini" file contains the two lines: hide_headers = False and hide_pcb_info = False, however, setting the values to "True" does not work. I.e hide_pcb_info = True does not work.

peter-jensen, 2019-09-27:
"hide_pcb_info = 1" works, However "group_connectors = 0" does not work.

Halbelunge, 2020-06-08:
Is there any way to deactivate "group_connectors" at the moment?
I get understable errors because to parts with the same footprint have different descriptions.
Is there any workaround for that?

Also it seems, that everytime I start it, it generates a new bom.ini file

set-soft, 2020-07-20:
@schneidersoft solution is correct.
These options has some inconsistencies in the write part, that's why the are "False" and not "0".
I fix it and other incosistencies in: https://github.com/INTI-CMNB/KiBoM/pull/11
I'll PR this change when @SchrodingersGat merges my other PRs. 
This is because I also fix code that is inside the pending PRs (#112 in particular).

set-soft, 2020-07-20:
> Is there any way to deactivate "group_connectors" at the moment?

Did you try the following?

```
group_connectors = 0
```

In the current code anything different than '1', 'yes' or 'true' (case insensitive) will disable it.

> Also it seems, that everytime I start it, it generates a new bom.ini file

Yes, if **bom.ini** is absent is created again. Like some virus ;-)
I don't like it, but is part of the fact that the script is designed to work as a "magic" plug-in to KiCad.

Halbelunge, 2020-07-21:
Yes I tried exactly this. Had no impact, at all.

Halbelunge, 2020-07-21:
[image]

What I did, was this in the end, So it stops the grouping, if it starts with a "J" what indicates most of times connectors.

Not pretty, but worked for the moment :)

set-soft, 2020-07-21:
Do you have a small example to reproduce this problem?
I failed using a trivial example.

I used the following netlist:  [connectors.xml](https://github.com/INTI-CMNB/KiBoM/blob/master/tests/input_samples/connectors.xml)

It contains 4 connectors with 4 different names, but using 2 different footprints.

Then I tried without a configuration file and got:
```
Component,Description,Part,References,Value,Footprint,Quantity Per PCB,Datasheet
1,"Generic connector, single row, 01x06, script generated (kicad-library-utils/schlib/autogen/connector/)",Conn_01x06_Male,J1 J2,Bizarre Bus,PinHeader_1x06_P2.54mm_Vertical,2,~
2,"Generic connector, single row, 01x04, script generated (kicad-library-utils/schlib/autogen/connector/)",Conn_01x04_Female,J3 J4,Bla bla connector,PinHeader_1x04_P2.54mm_Horizontal,2,~
```

Then I tried with the following configuration file: [no_group_connectors.ini](https://github.com/INTI-CMNB/KiBoM/blob/master/tests/config_samples/no_group_connectors.ini)
And got:
```
Component,Description,Part,References,Value,Footprint,Quantity Per PCB,Datasheet
1,"Generic connector, single row, 01x04, script generated (kicad-library-utils/schlib/autogen/connector/)",Conn_01x04_Female,J4,Another connector,PinHeader_1x04_P2.54mm_Horizontal,1,~
2,"Generic connector, single row, 01x06, script generated (kicad-library-utils/schlib/autogen/connector/)",Conn_01x06_Male,J1,Bizarre Bus,PinHeader_1x06_P2.54mm_Vertical,1,~
3,"Generic connector, single row, 01x04, script generated (kicad-library-utils/schlib/autogen/connector/)",Conn_01x04_Female,J3,Bla bla connector,PinHeader_1x04_P2.54mm_Horizontal,1,~
4,"Generic connector, single row, 01x06, script generated (kicad-library-utils/schlib/autogen/connector/)",Conn_01x06_Male,J2,Output crazy,PinHeader_1x06_P2.54mm_Vertical,1,~
```

You can try them in my [fork](https://github.com/INTI-CMNB/KiBoM/) running:
```.{bash}
make test_local
```
You'll  get an `output` directory containing the results of the tests in `GroupConnectors` and `NoGroupConnectors` subdirs.

BTW, the results of the current test can be downloaded from the GitHub Actions, i.e. [Test_Output.zip](https://github.com/INTI-CMNB/KiBoM/suites/946200848/artifacts/11747970)

Halbelunge, 2020-07-24:
So this is the thing that I exportet via the BOM dialog in KiCAD. In the bom.ini is 

; If 'group_connectors' option is set to 1, connectors with the same footprints will be grouped together, independent of the name of the connector
group_connectors = 0

So it should not group the connectors.
If you look in the .html at position 7, you see that J1 J6 J11 are grouped, even if they should not.
Is used your stock version. I am not sure how I parse your test xml from Kicad.
[OpenBikeSensor_pcb.zip](https://github.com/SchrodingersGat/KiBoM/files/4974054/OpenBikeSensor_pcb.zip)

Halbelunge, 2020-07-24:
Ahh I see what the difference is. I have for all three connectors the same value. So it does not split them because of this. 

I dont know if this is intended. But I think it should decide it on the designator "J" or rather on a set of 
configurable designators.

set-soft, 2020-07-24:
The `[GROUP_FIELDS]` section controls how to group components.

Common practice is to use the *Value* field of connectors to describe what they are i.e. "Power"

This is why we need the `group_connectors` option, so they are grouped. If you want to separate them put the comment (bemerkung?) information in the *Value*. Or use another grouping criteria (i.e. the manufacturer part number).