SchrodingersGat/KiBoM #69: Generate the netlist in xml format

joel-felcana, 2019-10-11:
Hi,

I'm trying to run KiBom from the command line with no GUI interaction at all. I noticed that in projects where I haven't run the BOM from the GUI, there's no xml netlist (at least on windows with Kicad 5.1.4).

Is there a way to generate that netlist so I can run KiBom from the command line? The alternative I see is runing the BoM at least once from the GUI, but that's prone to errors as it would be very easy to forget to update the XML if something changes in the schematic, and you would be generating an outdated BoM

KarlZeilhofer, 2019-11-19:
I have a very similar problem here. 
I even cannot generate a XML netlist with the GUI. Also I cannot run KiBOM from the GUI - I think this is related to the AppImage I'm using. 

I'm running KiCad 5.1.4 on Linux Mint 19.2.

KarlZeilhofer, 2019-11-20:
Just want to link to my topic in the KiCad forum here: https://forum.kicad.info/t/confusion-about-running-kibom-in-eeschema/19921

SchrodingersGat, 2019-11-20:
I do not think that these issues are related to KiBOM - rather the initial netlist generation step. KiBOM uses the .xml netlist file which must be first generated by KiCad

@joel-felcana I do not know how to generate the netlist from the command line. I would imagine there is a way to do it?

joel-felcana, 2019-11-26:
I've done a bit of research and you can generate an XML with a python library called skidl (https://xesscorp.github.io/skidl/docs/_site/index.html) but you still need a netlist file to do it, which can't be generated via command line either. Normally my netlists are more up to date than my XMLs, so it's a step forward, but it's still non-ideal.
Given that python support for eeschema won't be rolled out until kicad6, I think I'll park it for now.

Thanks!

closed 2019-11-26 by joel-felcana