SchrodingersGat/KiBoM #70: Workaround for broken Nightly - ConfigParser.NoOptionError

MichaelRColton, 2019-12-16:
At some point in the Nightly builds of KiCAD, something changed that seems to have broken KiBoM. I was getting ConfigParser.NoOptionErrors in two places.

I was able to work around the issue (in an inelegant way) by changing lines 143-144 preferences.py in bomlib to:

            self.outputFileName = "bom"
            self.variantFileNameFormat = "csv"

SchrodingersGat, 2019-12-17:
@MichaelRColton can you raise a PR with this fix? It seems like a reasonable solution :)

MichaelRColton, 2019-12-17:
Is it? I don't really understand the original code well enough, so I just hard coded values that allowed me to get a Bom exported, but maybe we lose the ability to change the file name and type by doing this. More than anything, opening this issue was meant to be a heads up and a temporary  work around for other people experiencing the issue.

(It's not that I'm unwilling to do more, I just don't think I am really up to doing it right)

bootchk, 2020-02-12:
This issue still exists:

In the KiCad output, get:

`Run command:
python "/home/bootch/Downloads/KiBoM-master/KiBOM_CLI.py" "/home/bootch/Dropbox/solarmobile/PCBoards/BLDC/BQ/BQNidecr2/BLDC.xml" "/home/bootch/Dropbox/solarmobile/PCBoards/BLDC/BQ/BQNidecr2/BLDC.html"

Command error. Return code 1
Error messages:
Traceback (most recent call last):
  File "/home/bootch/Downloads/KiBoM-master/KiBOM_CLI.py", line 186, in <module>
    pref.Read(config_file)
  File "/home/bootch/Downloads/KiBoM-master/bomlib/preferences.py", line 143, in Read
    self.outputFileName = cf.get(self.SECTION_GENERAL, self.OPT_OUTPUT_FILE_NAME)
  File "/usr/lib/python2.7/ConfigParser.py", line 340, in get
    raise NoOptionError(option, section)
ConfigParser.NoOptionError: No option u'output_file_name' in section: u'BOM_OPTIONS'`

I am not complaining, just adding info.  I haven't tried the workaround or looked at the code.

I did not create the bom.ini myself, KiBOM seems to have created it.  I did look at the bom.ini and indeed there does not seem to be a 'output_file_name' field.

bootchk, 2020-02-12:
Just a quick look at the commit in schneidersoft's clone seems like it might be more elegant.  But I don't have deep knowledge about the code.

bootchk, 2020-02-12:
But the fix in schneidersoft's clone also throws an exception: 

TypeError: get() got an unexpected keyword argument 'default'

bootchk, 2020-04-12:
Now, schneidersoft's clone seems to have fixed this issue, and this repository seems to still have this issue.  Schneidersoft's code seems to have significantly changed preferences.py.

This repository seems to have made significant changes to ease installation (using pip instead of downloading and unzipping an archive.)

Ideally, the repositories should be merged.

Thanks to all parties.

SchrodingersGat, 2020-07-19:
This should now have been fixed in https://github.com/SchrodingersGat/KiBoM/pull/121

closed 2020-07-19 by SchrodingersGat