SchrodingersGat/KiBoM #94: fit_field and output subdirectory problem

amat90, 2020-03-23:
I'm using PIP version 1.7.0 on linux. I found two issues:

1. I specified fit_field = VariantPCB. When I generate BOM files and this parameter is:
- compleatly empty - Quantity = 1 - that's ok
- any character inside, it could be anything: "" or - or a - Quantity = 1(DNC)
- DNF - Quantity = 1(DNF)(DNC)
- parameters of PCB variants doesn't work too. "-VAR1" or "+VAR2" gives Quantity = 1(DNC)

2. In PIP version doesn't work putting output files to specified folder (command line -d BOM). Files are generated correctly (in my test project: temp_BOM_rev.85_2.html), there is generated new folder, with name temp.html, which is empty. There's no folder "BOM".
In previous version, clon from github it was working fine.

One more thing about linux version of KiCad. There is no possibility to add only the command line in BOM script. Program needs to get file, and then gives access to edit and run " python3 -m kibom "%I" "%O" ".
I override this by creating a new .py file:

```
#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
@package

KiBOM - Bill of Materials generation for KiCad

Command line:
	python3 -m kibom "%I" "%O.csv" -s ; -d BOM -n 1 -r default

	"%O" - output format
		- "%O.csv"
		- "%O.html"

	-s - column separator
	-d - output subdirectory
	-n - number of PCB to calculate
	-r - variant of PCB 
"""
print("\n\nTO GENERATE OUTPUT FILES USE COMMAND LINE FROM DESCRIPTION")
```

My test command line:
python3 -m kibom "%I" "%O.html" -s ; -d BOM -n 1

Regards

amat90, 2020-05-03:
I found one more problem. Components on PCB with Fabrication Attribute set to "Virtual" are visible in BOM.

SchrodingersGat, 2020-05-05:
> In PIP version doesn't work putting output files to specified folder (command line -d BOM). Files are generated correctly (in my test project: temp_BOM_rev.85_2.html), there is generated new folder, with name temp.html, which is empty. There's no folder "BOM".
In previous version, clon from github it was working fine.

This issue is being fixed in https://github.com/SchrodingersGat/KiBoM/pull/99

> I found one more problem. Components on PCB with Fabrication Attribute set to "Virtual" are visible in BOM.

KiBoM does not look at the PCB file to determine which parts are "virtual". Perhaps this could be a future improvement.

amat90, 2020-05-06:
arek@Aspire-5742G:~$ pip3 install kibom -U
Collecting kibom
  Using cached https://files.pythonhosted.org/packages/ae/62/ca12e16ddbd9b3f6abe75de51916afa9fe8b4a6e7f8ee95b967a082dec29/kibom-1.7.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-h7wwky3k/kibom/

SchrodingersGat, 2020-05-06:
Weird - might this be related to this issue here - https://github.com/pypa/pip/issues/1064 ?

I have just tried the following:

`pip install kibom -U`
`pip pip3 install https://files.pythonhosted.org/packages/ae/62/ca12e16ddbd9b3f6abe75de51916afa9fe8b4a6e7f8ee95b967a082dec29/kibom-1.7.1.tar.gz` 

on both windows and linux. All seems tow work.

amat90, 2020-05-06:
OK, i first installed:
`pip3 install setuptools`
Then:
`pip3 install kibom -U`
and I get:
`Collecting kibom
  Using cached https://files.pythonhosted.org/packages/ae/62/ca12e16ddbd9b3f6abe75de51916afa9fe8b4a6e7f8ee95b967a082dec29/kibom-1.7.1.tar.gz
Collecting xlsxwriter (from kibom)
  Using cached https://files.pythonhosted.org/packages/00/1f/2092a81056d36c1b6651a645aa84c1f76bcee03103072d4fe1cb58501d69/XlsxWriter-1.2.8-py2.py3-none-any.whl
Building wheels for collected packages: kibom
  Running setup.py bdist_wheel for kibom ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-dy0fsq2j/kibom/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpbhz413expip-wheel- --python-tag cp36:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for kibom
  Running setup.py clean for kibom
Failed to build kibom
Installing collected packages: xlsxwriter, kibom
  Running setup.py install for kibom ... done
Successfully installed kibom-1.7.1 xlsxwriter-1.2.8`

There are some errors, but when I check version it is updated:
`python3 -m kibom --version
KiBOM Version: 1.7.1
`
Putting files into folder works fine now, thanks ;D

PS: Problem with DNC  still exist... ;)