SchrodingersGat/KiBoM #97: Command error. Return code 1 [KiCad 5.1.5] [MacOS 10.15.4]

hofstragroup, 2020-04-23:
I'm trying to set up kibom for the first time. I want to run it from within kicad Bill of Material window. The first issue I ran into is that, contrary to readme, I could not find any file "KiBOM_CLI.py". So instead I selected __main__.py when adding the kibom plugin.

When I run python `"/Users/user1/Library/Preferences/kicad/scripting/KiBoM/kibom/__main__.py" "%I" "%O"` I get `Command error. Return code 1
Error messages:
Traceback (most recent call last):
  File “/Users/peer/Library/Preferences/kicad/scripting/KiBoM/kibom/__main__.py”, line 26, in <module>
    from .columns import ColumnList
ValueError: Attempted relative import in non-package` and no output

If I instead add the plugin by selecting __init__.py and running `"/Users/peer/Library/Preferences/kicad/scripting/KiBoM/kibom/__init__.py" "%I" "%O` I see a success message and eeschema xml netlist created in project directory.

Please help me set up KiBoM.

SchrodingersGat, 2020-04-25:
The `KiBoM_CLI.py` script is now outdated, please refer to the usage details here:  https://github.com/SchrodingersGat/KiBoM/blob/master/README.md#usage

Hope this helps! LMK if you still have issues.

hofstragroup, 2020-05-06:
Thanks for responding. I did attempt the method suggested in the readme, namely installing the script via pip and then running via kicad with `python -m kibom "%I" "%O.csv"`. There are a couple problems with this on macOS. The first is that the macOS version of kicad requires that a file be selected when adding a BOM plugin. I got around this by selecting a dummy file and then typing `python -m kibom "%I" "%O.csv"` in the BOM plugin command line. The second problem is I have been unable to get kibom installed for the same version of python that kicad uses.

I believe that part of the difficulty relates to kicad on macOS using python 2.7 which is deprecated. So, while `pip3 install kibom` succeeds, `pip install kibom` fails. I then tried `python2.7 -m pip install kibom` which also fails with an error associated with md5 and openssl. After upgrading openssl and importing hashlib to get past the md5 issue, I got `python2.7 -m pip install kibom`to run successfully, but this still doesn't associate kibom with the correct version of python.

I know from `python –version` within kicad that kicad is using python2.7.16, but I believe that `python2.7 -m pip install kibom` is installing kibom for python2.7.15

How should I install kibom for the same version of python being used by kicad?