SchrodingersGat/KiBoM #98: Kibom bom folder -d argument not working.

OneAudio, 2020-05-02:
Hello,
First, thank you and congratulation for your great work !

I use Kibom on Ubuntu (16.04) and Python 2.7.

I'm not able to use -d argument to  create the bom file in a specific
project folder.
I use the command : python -m kibom -d bom "%I" "%O.csv"
The directory name is "bom".

I installed version with pip.
I tried also latest version 1.7.1pre but there is same problem.
Thank you very much again !

Frex

SchrodingersGat, 2020-05-04:
This feature is working for me. Can you please run the script with the `-vvv` option (for full verbose output) and provide the output of the script? Thanks

OneAudio, 2020-05-05:
HEllo,

Below the result of the scrip with 1.7.1pre version(latest).
Best regards.

Frex

Run command:
python -m kibom -vvv -d bom "/home/olivier/PROJECTS/Git/AA10M08/AA10M08.xml" "/home/olivier/PROJECTS/Git/AA10M08/AA10M08.csv"

Success
Info messages:
 KiBOM version 1.7.1pre
 Input: /home/olivier/PROJECTS/Git/AA10M08/AA10M08.xml
 Configuration file: /home/olivier/PROJECTS/Git/AA10M08/bom.ini
 PCB variant: [u'default']
INFO Excluding 'H1': Field 'Part' (MountingHole_Pad) matched 'mount.*hole'
INFO Excluding 'H2': Field 'Part' (MountingHole_Pad) matched 'mount.*hole'
INFO Excluding 'H3': Field 'Part' (MountingHole_Pad) matched 'mount.*hole'
INFO Excluding 'H4': Field 'Part' (MountingHole_Pad) matched 'mount.*hole'
INFO Excluding 'H5': Field 'Part' (MountingHole_Pad) matched 'mount.*hole'
INFO Excluding 'H6': Field 'Part' (MountingHole_Pad) matched 'mount.*hole'
INFO Excluding 'TP1': Field 'References' (TP1) matched '^TP[0-9]*'
INFO Excluding 'TP10': Field 'References' (TP10) matched '^TP[0-9]*'
INFO Excluding 'TP11': Field 'References' (TP11) matched '^TP[0-9]*'
INFO Excluding 'TP12': Field 'References' (TP12) matched '^TP[0-9]*'
INFO Excluding 'TP13': Field 'References' (TP13) matched '^TP[0-9]*'
INFO Excluding 'TP14': Field 'References' (TP14) matched '^TP[0-9]*'
INFO Excluding 'TP15': Field 'References' (TP15) matched '^TP[0-9]*'
INFO Excluding 'TP16': Field 'References' (TP16) matched '^TP[0-9]*'
INFO Excluding 'TP17': Field 'References' (TP17) matched '^TP[0-9]*'
INFO Excluding 'TP18': Field 'References' (TP18) matched '^TP[0-9]*'
INFO Excluding 'TP19': Field 'References' (TP19) matched '^TP[0-9]*'
INFO Excluding 'TP2': Field 'References' (TP2) matched '^TP[0-9]*'
INFO Excluding 'TP20': Field 'References' (TP20) matched '^TP[0-9]*'
INFO Excluding 'TP21': Field 'References' (TP21) matched '^TP[0-9]*'
INFO Excluding 'TP22': Field 'References' (TP22) matched '^TP[0-9]*'
INFO Excluding 'TP23': Field 'References' (TP23) matched '^TP[0-9]*'
INFO Excluding 'TP24': Field 'References' (TP24) matched '^TP[0-9]*'
INFO Excluding 'TP25': Field 'References' (TP25) matched '^TP[0-9]*'
INFO Excluding 'TP26': Field 'References' (TP26) matched '^TP[0-9]*'
INFO Excluding 'TP27': Field 'References' (TP27) matched '^TP[0-9]*'
INFO Excluding 'TP28': Field 'References' (TP28) matched '^TP[0-9]*'
INFO Excluding 'TP29': Field 'References' (TP29) matched '^TP[0-9]*'
INFO Excluding 'TP3': Field 'References' (TP3) matched '^TP[0-9]*'
INFO Excluding 'TP30': Field 'References' (TP30) matched '^TP[0-9]*'
INFO Excluding 'TP31': Field 'References' (TP31) matched '^TP[0-9]*'
INFO Excluding 'TP32': Field 'References' (TP32) matched '^TP[0-9]*'
INFO Excluding 'TP33': Field 'References' (TP33) matched '^TP[0-9]*'
INFO Excluding 'TP34': Field 'References' (TP34) matched '^TP[0-9]*'
INFO Excluding 'TP35': Field 'References' (TP35) matched '^TP[0-9]*'
INFO Excluding 'TP36': Field 'References' (TP36) matched '^TP[0-9]*'
INFO Excluding 'TP37': Field 'References' (TP37) matched '^TP[0-9]*'
INFO Excluding 'TP38': Field 'References' (TP38) matched '^TP[0-9]*'
INFO Excluding 'TP39': Field 'References' (TP39) matched '^TP[0-9]*'
INFO Excluding 'TP4': Field 'References' (TP4) matched '^TP[0-9]*'
INFO Excluding 'TP40': Field 'References' (TP40) matched '^TP[0-9]*'
INFO Excluding 'TP41': Field 'References' (TP41) matched '^TP[0-9]*'
INFO Excluding 'TP5': Field 'References' (TP5) matched '^TP[0-9]*'
INFO Excluding 'TP6': Field 'References' (TP6) matched '^TP[0-9]*'
INFO Excluding 'TP7': Field 'References' (TP7) matched '^TP[0-9]*'
INFO Excluding 'TP8': Field 'References' (TP8) matched '^TP[0-9]*'
INFO Excluding 'TP9': Field 'References' (TP9) matched '^TP[0-9]*'
WARNING Field conflict: (R25 R26 R43 R44 R46 R47) [Mfg Ref] : 'CRCW0603200RFKEAC' <- 'CRCW06030000Z0EAC'
WARNING Field conflict: (R25 R26 R43 R44 R46 R47) [Order code] : '71-CRCW0603200RFKEAC' <- '71-CRCW06030000Z0EAC'
WARNING Field conflict: (R25 R26 R43 R44 R46 R47) [Part Description] : 'Thick Film Resistors - SMD 1/10Watt 200ohms 1% Commercial Use' <- 'Thick Film Resistors - SMD 1/10Watt 0ohms Commercial Use'
WARNING Field conflict: (R25 R26 R43 R44 R46 R47) [Distributor] : 'Mouser' <- '~'
WARNING Field conflict: (R25 R26 R43 R44 R46 R47) [Mfg Name] : 'Vishay' <- '~'
WARNING Field conflict: (R25 R26 R43 R44 R46 R47) [Mfg Ref] : 'CRCW0603200RFKEAC CRCW06030000Z0EAC' <- '~'
WARNING Field conflict: (R25 R26 R43 R44 R46 R47) [Order code] : '71-CRCW0603200RFKEAC 71-CRCW06030000Z0EAC' <- '~'
WARNING Field conflict: (R25 R26 R43 R44 R46 R47) [Part Description] : 'Thick Film Resistors - SMD 1/10Watt 200ohms 1% Commercial Use Thick Film Resistors - SMD 1/10Watt 0ohms Commercial Use' <- '~'
WARNING Field conflict: (R25 R26 R43 R44 R46 R47) [Unit price] : '0,09' <- '~'
WARNING Field conflict: (R35 R45) [Mfg Ref] : 'CRCW06030000Z0EAC' <- 'CRCW060333R0FKEAC'
WARNING Field conflict: (R35 R45) [Order code] : '71-CRCW06030000Z0EAC' <- '71-CRCW060333R0FKEAC'
WARNING Field conflict: (R35 R45) [Part Description] : 'Thick Film Resistors - SMD 1/10Watt 0ohms Commercial Use' <- 'Thick Film Resistors 1/10Watt 33ohms 1% Commercial Use'
INFO Removing: Build Quantity
 Output: /home/olivier/PROJECTS/Git/AA10M08/AA10M08_bom_1.0.csv
INFO CSV Output -> /home/olivier/PROJECTS/Git/AA10M08/AA10M08_bom_1.0.csv

.

SchrodingersGat, 2020-05-05:
I think I have fixed this now!

closed 2020-05-05 by SchrodingersGat

OneAudio, 2020-05-07:
Hello,

I updated my files and you are right, that's work now. Great !
Thank you very much for this fast fix.

I use a bom directory (why i wanted to use -d ), and i would like to locate
the bom.ini file also in this folder (like the Kibom bom file).

As I understood the documentation of the command line,
i can use the --cfg argument.
SOo I tried to use this one :
python -m kibom -d bom --cfg bom\bom.ini "%I" "%O.html"

When i sent this command no error has been returned, but this is not
my bom.ini file (in \bom folder)  that has been used to generate the bom
file by the scrip but the default on (I can see the wrong parts order).
It is a bug, or a miss-understanding of the command line from me ?

Best regards.

Frex