SchrodingersGat/KiBoM #102: Installing Kibom in windows

firebladed, 2020-06-17:
the readme for installing kibom in windows is unclear

the readme still mentions KiBOM_CLI.py in the quick start section

trying to use the 2.7 python from kicad appears not to work
`C:\Program Files\KiCad\bin>pip install kibom`
works 
however
`python -m kibom "%I" "%O"`
in the BOM window gets
`Error messages:
Traceback (most recent call last):
  File "C:\Program Files\KiCad\lib\python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\Program Files\KiCad\lib\python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Program Files\KiCad\lib\python2.7\site-packages\kibom/__main__.py", line 198, in <module>
    main()
  File "C:\Program Files\KiCad\lib\python2.7\site-packages\kibom/__main__.py", line 170, in main
    pref.Read(config_file)
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/kibom/preferences.py", line 143, in Read
    self.outputFileName = cf.get(self.SECTION_GENERAL, self.OPT_OUTPUT_FILE_NAME)
  File "C:\Program Files\KiCad\lib\python2.7/ConfigParser.py", line 340, in get
    raise NoOptionError(option, section)
ConfigParser.NoOptionError: No option u'output_file_name' in section: u'BOM_OPTIONS'`

trying with a separate install of python3 gets (with paths removed)
` 
C:\Users\fireblade>pip3 install kibom
Requirement already satisfied: kibom in c:\users\fireblade\appdata\local\programs\python\python38-32\lib\site-packages (1.7.1)
Requirement already satisfied: xlsxwriter in c:\users\fireblade\appdata\local\programs\python\python38-32\lib\site-packages (from kibom) (1.2.9)

C:\Users\fireblade>python -m kibom "<inpath>" "<outpath>"
 KiBOM version 1.7.1
 Output directory: '<outpath>'
 Input: <inpath>
Traceback (most recent call last):
  File "C:\Users\fireblade\AppData\Local\Programs\Python\Python38-32\lib\configparser.py", line 789, in get
    value = d[option]
  File "C:\Users\fireblade\AppData\Local\Programs\Python\Python38-32\lib\collections\__init__.py", line 898, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "C:\Users\fireblade\AppData\Local\Programs\Python\Python38-32\lib\collections\__init__.py", line 890, in __missing__
    raise KeyError(key)
KeyError: 'output_file_name'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\fireblade\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\fireblade\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\fireblade\AppData\Local\Programs\Python\Python38-32\lib\site-packages\kibom\__main__.py", line 198, in <module>
    main()
  File "C:\Users\fireblade\AppData\Local\Programs\Python\Python38-32\lib\site-packages\kibom\__main__.py", line 170, in main
    pref.Read(config_file)
  File "C:\Users\fireblade\AppData\Local\Programs\Python\Python38-32\lib\site-packages\kibom\preferences.py", line 143, in Read
    self.outputFileName = cf.get(self.SECTION_GENERAL, self.OPT_OUTPUT_FILE_NAME)
  File "C:\Users\fireblade\AppData\Local\Programs\Python\Python38-32\lib\configparser.py", line 792, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'output_file_name' in section: 'BOM_OPTIONS'
`

firebladed, 2020-06-17:
ok looks like a  result of having "output_file_name" missing from bom.ini

SchrodingersGat, 2020-07-18:
@firebladed I have now reverted to use the original `kibom_cli.py` script as it is *much* easier to support. Cheers

closed 2020-07-18 by SchrodingersGat