SchrodingersGat/KiBoM #144: fit_field not working on Win7

aiotech-pub, 2020-10-23:
I have a bom.ni local file where:

; Field name used to determine if a particular part is to be fitted
fit_field = FIT

then to test KiBoM I edited a capacitor (C4) on the schematic adding the FIT field filled with:
"+a,-b"

this is my command as typed in Kicad:
`python "C:\Dati\Kicad\scripts\KiBoM-master/KiBOM_CLI.py" "-r b" "%I" "%O.html"`

on the html file I would expect to don't see C4 listed, instead it is listed and the field on the FIT column contains "+a,-b"

this is my command as typed in Kicad:
`python "C:\Dati\Kicad\scripts\KiBoM-master/KiBOM_CLI.py" "-r a" "%I" "%O.html"`

on the html file I would expect to see C4 listed, it is listed and the field on the FIT column contains "+a,-b"

In general terms I would expect a different behaviour:
if -r present on the command row: in FIT only fields containing - and + will be considered and the output will be produced with or without them only reporting texts not containing - and + (for instance dnf), if all the fileds will be empty or no dnf (And similar) texts are not found the column will not br printed
if -r not present on the command row: only reporting texts not containing - and + (for instance dnf) if all the fileds will be empty or no dnf (And similar) texts are not found the column will not br printed

eeintech, 2020-12-15:
@aiotech-pub This is the list of values for the `FIT` field for deciding whether or not to populate a part:
``` python
# String matches for marking a component as "do not fit"
DNF = [
    "dnf",
    "dnl",
    "dnp",
    "do not fit",
    "do not place",
    "do not load",
    "nofit",
    "nostuff",
    "noplace",
    "noload",
    "not fitted",
    "not loaded",
    "not placed",
    "no stuff",
]
```

`+a,-b` does not match any of them therefore the parts are going to be **fitted**.

Note that the `-r` argument won't help you here, it is used to set the variants and I don't think there is any logic in place (similar to yours) to process those parts.

eeintech, 2020-12-15:
@aiotech-pub Have you tried this command instead:
```python "C:\Dati\Kicad\scripts\KiBoM-master/KiBOM_CLI.py" -r a "%I" "%O.html"```

It looks like there are two proposed fixes for the variant logic: #136 and #137
Could be related.

set-soft, 2020-12-29:
I'll suggest trying https://github.com/INTI-CMNB/KiBoM this fork has #137 fix