SchrodingersGat/KiBoM #146: configparser.NoOptionError for default generated bom.ini

hkennyv, 2020-11-16:
as the title says, the default bom.ini that is generated by kibom will raise a configparser.NoOptionError when you try to run it. the first exception that is raised is:

```
  File "/usr/local/opt/python@3.8/Frameworks/Python.framework/Versions/3.8/lib/python3.8/configparser.py", line 792, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'output_file_name' in section: 'BOM_OPTIONS'
```

and if you add the `variant_file_name_format` to the bom.ini and re-run it raises again:

```
  File "/usr/local/opt/python@3.8/Frameworks/Python.framework/Versions/3.8/lib/python3.8/configparser.py", line 792, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'variant_file_name_format' in section: 'BOM_OPTIONS'
```

it seems like there are a couple solutions:

- have the default bom set `output_file_name` and `variant_file_name_format` (not a good option because typically the user specifies the output file name in the CLI)
- mark both of these options as optional

I'll try to look into this later and put in a PR

set-soft, 2020-11-19:
Which version of KiBoM are you using? 
Is the last from GitHub?
This error should be fixed by #121 
Also related to this PR in my fork: https://github.com/INTI-CMNB/KiBoM/pull/11

hkennyv, 2020-11-19:
@set-soft is correct. i was using kibom v1.7.1 (which is the current version on pypi btw, can we get a new release?)

after cloning the repo and installing from the master branch, it works :+1:

closed 2020-11-19 by hkennyv