SchrodingersGat/KiBoM #147: in kibom.component.Component, self.libpart is always None

hkennyv, 2020-11-16:
I ran into this error:

```
  File "/Users/khuynh/me/rejoule/kicad/dcdc_mgthv/venv/lib/python3.8/site-packages/kibom/component.py", line 191, in getDescription
    ret = self.libpart.getDescription()
AttributeError: 'NoneType' object has no attribute 'getDescription'
```

i tried to trace it down and i found that in the constructor, `self.libpart` is initialized to `None`:

https://github.com/SchrodingersGat/KiBoM/blob/0ce83c2549eb303405c63a1f9538d42ebd75dbb5/kibom/component.py#L46-L48

This is particularly problematic because there are multiple places in this file that call some method on `self.libpart`, which is guaranteed to throw an AttributeError.

Particularly, the one I ran into is this path that gets executed and there's not guard for checking if `self.libpart is None`:

https://github.com/SchrodingersGat/KiBoM/blob/0ce83c2549eb303405c63a1f9538d42ebd75dbb5/kibom/component.py#L186-L200

A quick search for `self.libpart` shows that not all method calls have a guard to check if `self.libpart` is not None

It looks like `Component.setLibPart` gets called in netlist_reader.py, but for some reason, `self.libpart` was still None in my example. It seems like the solution is to ensure that there is a guard every time that a method on `self.libpart` is called

set-soft, 2020-11-19:
Note that the line 191 in the code you quote can't raise this exception.
I think you are using older code.
The `ret = self.libpart.getDescription()` is in line 195 and inside a `try` block. This should be catched by `except AttributeError`.

hkennyv, 2020-11-19:
good catch @set-soft, i _think_ i used the pypi installation. let me try installing from the repo.

hkennyv, 2020-11-19:
so i was able to more-or-less fix it, thanks @set-soft . There was a misplaced parenthesis on line 199 that I had to fix as well before I could get it to run and raise the AttributeError properly. 

After fixing that and getting it to run, I had a question. Is there a reason we cannot return `""` as the description here? It's telling me that it cannot get a description for a part, and upon looking through my xml netlist, I have quite a few components that have something like:

```
      <libsource lib="" part="Rsense" description=""/>
```

Which is what I am assuming is causing this error to be raised. So I'm wondering what the rationale is for raising an `AttributeError` instead of using `""` as the description and moving on.

https://github.com/SchrodingersGat/KiBoM/blob/0ce83c2549eb303405c63a1f9538d42ebd75dbb5/kibom/component.py#L193-L200

eeintech, 2020-12-15:
@hkennyv Assuming the `Description` field is set for all your parts in your library, try to update your eeschema parts. If it still does not work then you may have to remove/re-add them to your schematic.

[image]

gbalke, 2021-08-18:
I was getting the same error:
```
Traceback (most recent call last):                                                                  
  File "/home/greg/.local/lib/python3.8/site-packages/kibom/component.py", line 427, in getDatasheet
    ret = self.libpart.getDescription()                                                             
AttributeError: 'NoneType' object has no attribute 'getDescription'                                 
```
I was able to resolve this error by manually adjusting the `.xml` part file. Using the code snippet suggested by @hkennyv, I found that kicad was generating descriptions such as:
```xml
<libsource lib="" part="Device:R_Small_11" description="Resistor, small symbol"/>
```
This should be:
```xml
<libsource lib="Device" part="R_Small_11" description="Resistor, small symbol"/>
```
After the change, I no longer get the error.

This snippet of regex might help those in need as there were a lot of mistakes in my file:
```regex
:%s/lib=""\ part="\(.*\):\(.*\)"/lib="\1"\ part="\2"/g
```

I'm running on nightly so this might just be one of the latest bugs but I thought I should mention it should anyone run into it as well!

set-soft, 2021-08-19:
Hi @gbalke !
Which KiBoM version are you using? Is the current git code or older? I think this problem is already solved in the current code (as I mention above).

gbalke, 2021-08-27:
You're correct, it was the older version on PyPi. It was interesting to see KiCAD generating these sorts of xmls though :thinking:
Simply throwing an error doesn't necessarily solve the problem, just ignores it?

set-soft, 2021-08-30:
Hi @gbalke !
I think the problem is just a change between KiCad 4.x and 5.x XML format.
With the current code you should get the descriptions without problems. 
I don't think KiCad can generate XMLs that rises the current exception, not at least under normal operation. Do you have a valid XML that raises the error in the current code?