SchrodingersGat/KiBoM #149: Separating components with the same value/footprint but different attributes

apcountryman, 2020-12-04:
Operating System: Debian 10 (Buster)

KiCad eeschema version: 5.1.8+dfsg1-1~bpo10+1, release build (from Buster backports)

KiBom version: current master (baceef96e83815e769303577cb4c4d435712aa0d)

Capacitors that have the same value, footprint, and manufacturer (captured in a "Manufacturer" field); but different part tolerances (captured in a "Tolerance" field), ratings (captured in a "Rating" field, and part numbers (captured in a "Part Number" field are being used in a project as shown in the following table:

| Manufacturer | Part Number | Reference | Footprint | Value | Tolerance | Rating |
| :---: | :---: | :---: | :---: | :---: | :---: | :---: |
| KEMET | C0603C104K8RACTU | C1 | C_0603_1608Metric | 0.1uF | 10% | 10 V |
| KEMET | C0603C104M5RACTU | C2 | C_0603_1608Metric | 0.1uF | 20% | 50 V |
| KEMET | C0603C104M5RACTU | C3 | C_0603_1608Metric | 0.1uF | 20% | 50 V |

When the BOM is generated, all these parts end up in the same BOM row since they all have the same value and footprint:

| Manufacturer | Part Number | References | Footprint | Value | Tolerance | Rating | Quantity Per PCB |
| :---: | :---: | :---: | :---: | :---: | :---: | :---: | :---: |
| KEMET | C0603C104K8RACTU C0603C104M5RACTU | C1-C3 | C_0603_1608Metric | 0.1uF | 10% 20% | 10 V 50 V | 3 |

I thought using a `[JOIN]` entry in the configuration file to combine the "Value", "Tolerance", and "Rating" fields might create separate rows but it appears that the fields are joined after the rows have already been combined:

```
[JOIN]
Value	Tolerance	Rating
```
| Manufacturer | Part Number | References | Footprint | Value | Quantity Per PCB |
| :---: | :---: | :---: | :---: | :---: | :---: |
| KEMET | C0603C104K8RACTU C0603C104M5RACTU | C1-C3 | C_0603_1608Metric | 0.1uF 10% 20% 10 V 50 V | 3 |

I've reviewed the README, and opened and closed issues but did not see anything other than `[JOIN]` that possibly addressed separating components with the same value/footprint but different attributes. Is this behavior that KiBoM supports? Is `[JOIN]` working as intended? I can't provide the actual KiCad project and KiBoM configuration, but could put together a minimal project that reproduces this behavior if that would be valuable. I'm currently not familiar enough with Python to look into this and supply a pull request, but would be able to sometime in the future.

This can be worked around for now by including the attributes in the value (which could improve schematic readability), or using a different footprint when the attributes vary.

set-soft, 2020-12-04:
You are right about "Join" is applied after grouping.

Did you try adding Tolerance to the GROUP_FIELDS section?

I usually group using the part number, to be more specific the Digikey part code. Of course it means all the components must have it, even if they are provided by another distributor.

apcountryman, 2020-12-04:
> You are right about "Join" is applied after grouping.
> 
> Did you try adding Tolerance to the GROUP_FIELDS section?
> 
> I usually group using the part number, to be more specific the Digikey part code. Of course it means all the components must have it, even if they are provided by another distributor.

Adding the `Tolerance` and `Rating` fields to the `[GROUP_FIELDS]` section did exactly what I was looking for. I'll probably add the `Manufacturer` and `Part Number` fields as well.

Thank you very much for the assistance.

closed 2020-12-04 by apcountryman