SchrodingersGat/KiBoM #153: Error writing XLSX output in KiBoM 1.8.0

KKlim, 2021-02-06:
The script starts but no xlsx output file is generated. Some bugs are reported.
Other formats such as csv, tsv, txt, html work rather ok and generate output files.

Windows 10 Pro 64bit PL
KiCad version: 5.1.9

Run command:
python "C:\_KiCad\KiBoM/KiBOM_CLI.py" -v "C:\_KiCad\KiBoM\test\kibom-test.xml" "C:/_KiCad/KiBoM/test/kibom-test.xlsx"

Command error. Return code -1
Info messages:
 KiBOM version 1.8.0
 Output directory: 'C:\_KiCad\KiBoM\test'
 Input: C:\_KiCad\KiBoM\test\kibom-test.xml
 Configuration file: C:\_KiCad\KiBoM\test\bom.ini
 PCB variant: default
 Saving BOM File: C:\_KiCad\KiBoM\test\kibom-test_bom_A.xlsx
ERROR Error writing XLSX output
ERROR Error writing variant 'None'

set-soft, 2021-02-06:
Do you have the `xlsxwriter` python module installed?

KKlim, 2021-02-06:
I didn't install anything myself.  How should I install it?

KKlim, 2021-02-08:
After installing the XlsxWriter module, xlsx files are generated. 
Thanks for the hint. :-)

I have suggestions that in the documentation, instead of:
XLSX (Needs XlsxWriter Python module)
it was written:
XLSX (Needs XlsxWriter Python module to be installed).

Link to the description of how to install XlsxWriter:
https://xlsxwriter.readthedocs.io/getting_started.html

KKlim, 2021-02-09:
See also: https://github.com/SchrodingersGat/KiBoM/issues/143

closed 2021-02-09 by KKlim