SchrodingersGat/KiBoM #156: Is sorting lines by reference possible?

BillyPocketNC, 2021-05-27:
Is it possible to sort the BOM lines by the Reference (first reference in the reference cell)?

I would like to have everything sorted by the reference.
This is a screen shot of what I am getting so far.
[image]
This is what I want to get.
[image]

Thanks for everyone's work on this project.

set-soft, 2021-05-27:
I think you really want your row 8 (C7) after the row 1 (C1). And J3 after J1, but before J120, etc.

BillyPocketNC, 2021-05-27:
@set-soft You are correct, I forgot to tell Libre Office Calc to use Natural sort.
Following is the corrected order:
[image]

set-soft, 2021-05-27:
I think KiBoM only sorts like this:

1. The type of component (C* before R*)
2. The component value (22 pF before 1 nF)

Then inside the same row, references are sorted in "natural order". But you can't sort the rows by reference.

set-soft, 2021-05-27:
I commented it in the PR: But this should be controlled by an option!

In fact I like the current behavior and not this one. BTW: you don't need an extra module, take a look at `kibom/sort.py` this is currently used to sort the refs inside the row. What you want is to apply it between rows. But again: only when the user asks for this. Some option, I don't know, may be `sort_by_ref` should control it.

BillyPocketNC, 2021-05-27:
I will take a look. sort_by_ref makes sense.

Now I see why you like the other sort method.
Thanks for the help @set-soft this was my first pull request.

SchrodingersGat, 2021-06-03:
@BillyPocketNC can you achieve the sorting you want here without using the `natsort` library?

BillyPocketNC, 2021-06-03:
@SchrodingersGat possibly, I haven't had a chance to try yet.