SchrodingersGat/KiBoM #158: Delete or sanitize generated XML netlist

ptr727, 2021-09-20:
Hi, new user, thank you.
I am running BOM tool from KiCad 5.99, and specifying `-d manufacture` as argument to create BOM in manufacture folder.
I noticed (in source control) that a new file projectname.xml was created, it appears to be the netlist in XML format, created by the BOM tool.
It is undesirable to have an interstitial file persisted in the project directory.

Consider instead:
- Deleting temporary / interstitial files after use.
- Creating the XML netlist in the same directory as the BOM, and naming it using the same convention as the BOM file.

set-soft, 2021-09-20:
Hi @ptr727. Isn't this the XML created by KiCad and used as input for KiBoM?

ptr727, 2021-09-20:
Yes, I assume it is part of asking KiCad to create a BOM in XML format, that is then used to create the BOM in KiBOM.
Regardless of "who" created the file, it was indirectly created by KiBOM, not by me, and is an undesirable file that showed up as part of KiBOM but not a KiBOM file.

In my mind it is thus an interstitial file, that should be deleted by KiBOM, or if deemed valuable, renamed and moved to the output folder.

ptr727, 2021-09-21:
Reading up, it seems the XML is generated by KiCad as part of just running the BOM tool, and not actually generated on command by KiBoM, so nothing to do in KiBoM.

closed 2021-09-21 by ptr727