SchrodingersGat/KiBoM #163: Missing or empty fields are not populated as empty and result in global BOM scrambling.

phansel, 2022-02-24:
Suppose you have a board with four parts: C1 (GCMxyz), D1 (BAS20), R1 (CRCWxyz), U1 (STM32F0). 

You do not specify the MPN of D1 for whatever reason. Or, perhaps D1 is a DNI and has no MPN assigned. 

If you group by MPN alone, KiBoM will generate a BOM with errors.

Expected BOM:

Line item   | MPN               | References
1                 GCMxyz          C1
2                                         D1
3                 CRCWxyz       R1
4                 STM32F0        U1           

Actual BOM generated:

Line item   | MPN               | References
1                 GCMxyz          C1
2                 CRCWxyz       D1
3                 STM32F0        R1
4                                          U1

You might hope that it leaves the unpopulated fields blank, rather than zero or (-1) length, but it does not. This off-by-one error can be ultra-subtle or even unnoticeable yet cause a board to be assembled completely wrong.

I'm not sure how to fix this issue.

SchrodingersGat, 2022-02-25:
@phansel in my initial investigation I am unable to replicate this.

Can you please provide a set of files as a minimal example?

- Kicad schematic file
- `bom.ini` configuration file
- command you are running to generate the BOM
- version of KiBOM

phansel, 2022-02-26:
I'll prep a set of demo files.

phansel, 2022-04-16:
Unable to reproduce this. :+1:

closed 2022-04-16 by phansel