SchrodingersGat/KiBoM #167: Generate BOM with all parts except those with 'dnp' in fit field

EeEmDee, 2022-03-29:
Hello there,

I want to generate a BOM where basically all parts are included, except those which are marked with 'dnp'.
As an example lets say we have some parts defined as follows:

| Part      | fit_field|
| ----------- | ----------- |
| R1      | dnp       |
| R2   |         |
| R3   | +16        |
| R4   | +16,+17        |
| R5   | -16        |
| R6   | dnp        |

In my BOM I would like to get R2, R3, R4 and R5. Only R1 and R6 would be left out.

As of now I tried to play with the config file, but there is only the option to include all parts and the dnp parts or to exclude anything, that doesn't have an empty fit_field.
I also wrote a script, where I had the idea to just use the -r option and put all available variants on the schematic in there, but then parts marked with a '-' would be also excluded. In the example above it would be R5.

What would be the a good way to get the desired behaviour?

Help would be appreciated :)

EeEmDee, 2022-03-29:
@SchrodingersGat Oh, I forgot to mention. The .ini file I am using has fit_field = fit_field, not like in the default configuration with fit_field=Config.

SchrodingersGat, 2022-03-29:
Ah, ok. Can you test with the default "Config" field? It should work, independent of what the field (column) is actually called, but perhaps there's a bug there...

EeEmDee, 2022-03-29:
@SchrodingersGat
In general all works as expected. I can create the correct BoMs for their respective variants and everything. I just want to create a BoM with all the parts, except where the fit_field (no matter its name) is 'dnp'. No matter the assigned variant. I hope you understand what I mean :)

set-soft, 2022-03-29:
Hi @EeEmDee !
Take a look at this file:

[kibom-variante.zip](https://github.com/SchrodingersGat/KiBoM/files/8373095/kibom-variante.zip)

Try: `KiBOM_CLI.py kibom-variante.xml pp.csv`

You'll see only R1, R2 and R3 are included. Most of the excluded components uses `Config=DNP` (tests various similar values DNF, Do Not Fit, etc.)

EeEmDee, 2022-03-31:
Hi @set-soft, same problem.

I have attached a minimal working example. There is no config file attached, so kibom will execute with default settings.

[example.zip](https://github.com/SchrodingersGat/KiBoM/files/8386984/example.zip)

When you execute `python "KiBOM_CLI.py" "%I" "%O"` it will generate a Netlist with only R3 in it.
But what I want is R2,R3,R4,R5.

set-soft, 2022-03-31:
Hi @EeEmDee !

Ok, now I see what you want.

You want to disable the variants mechanism and only filter by DNP.

Here are some ways to do it:

1. Use "DNP" in the value. Then create a .ini that defines something like `fit_field = None` and use it.
2. Add another field, lets say `DNP`. Make it `-1` for DNP components. Then define `fit_field = DNP`. And generate variant `1`

EeEmDee, 2022-04-01:
Hi @set-soft,

thanks for the response. The solutions seem reasonable, though for me they are not viable, because there is a certain infrastructure already built and if I would apply the changes as you proposed, then everything would go to hell lol.

I'm hoping that maybe the maintainer (@SchrodingersGat) could add the feature? Maybe as a setting in the config file. If not, then maybe I will fork the repo and code it myself.

set-soft, 2022-04-02:
Hi @EeEmDee !

If you are running KiCad on Linux, or you can use a CI/CD environment (like GitHub) you could try [KiBot](https://github.com/INTI-CMNB/KiBot), it implements KiBoM variants and you can configure filters to do what you want.

Running it on Windows is tricky (you must setup a Linux like environment).