SchrodingersGat/KiBoM #172: Column content is picked up from library, but not from Current Schematics

Tarik-T, 2022-10-11:
First of all, great tool. 
It's the only tool for me at the moment that provides a .csv file that I don't have to adapt to get a usable list in Excel
Thanks a lot for that!

However, the following problem:
In the library file (some .kicad_sym), description, datasheet, part number, order info, footprint, etc. are maintained (or not maintained).
In the schematic the information might be maintained differently, probably more up to date or at least more suitable for the current purposes.
My impression is that KiBOM pulls the information from the library file, which, to the extent that the more current information is in the schematic, leads to rework.  Assuming component information in the schematic is more current, I would prefer a hierarchy. 
First step, consider the details given to the symbol in the schematic, if there is nothing there, use the information from the library.
Is there a solution?

Thanks!
Tarik

SchrodingersGat, 2022-10-11:
@Tarik-T it should be giving preference to the schematic over the base library. That is my experience from regular use of KiBoM.

Can you provide a minimum working example showing this issue?

Tarik-T, 2022-10-12:
Hi,
Of Course
Here the schematics info
[image]

Here how it looks in library editor:
[image]
You might note, there is a "Schema" or a "lib" in the lines.

Result in BoM
[image]

The description does not seem to be from the schematic or the library.
My understanding of solution, there are several entries in the kicad_sch file. This is what is evaluated by KiBoM:
[image]

That is the one I m wondering if it could make more sense:
[image]

In summary, only the "Description" and "Part" fields are affected.
The description was used incorrectly by me.
Field "Part" ... I am not sure what is right or wrong.
Anyway, nothing more to complain.
Thanks for your patience - I ll close the ticket. 

Cheers,
Tarik

closed 2022-10-12 by Tarik-T

SchrodingersGat, 2022-10-17:
@Tarik-T you closed this out - did you get it working?

Tarik-T, 2022-10-20:
@SchrodingersGat 

Yes. 
Nothing to complain any more.
Thank you for that nice piece of software!

Tarik