SchrodingersGat/KiBoM #176: "Description" in BOM being pulled from symbol library description which is useless

ch992319, 2022-11-01:
I have noticed that when I generate a BOM with this program, it pulls from the symbol description of the symbol library. Typically I have a description in the symbol fields of the symbol of the schematic. Here is typically where I put tolerance, power dissipation, size, current, etc... usually pulling the Digikey Description off and pasting it to the description of the symbol, like "CAP CER 0.1UF 50V X7R 0603".

Instead the KiBOM program pulls from the symbol description from the library, something like "Unpolarized capacitor" or "Resistor, small US symbol" which quite frankly is useless.

To get around this, I went into the BOM.ini file and creates a new field named "Descriptions" with the added s at the end to differentiate, because I couldn't find anywhere to change where the "Description" is being pulled from. Cause if I have a field in the symbol called "Description" then it gets over written with the symbol library description and not the one I entered into the field on the schematic.

Can this be changed? Is there a way that I can change it? Also, is there a way to have an over arching bom.ini that gets pulled from each time instead of going with the default? I thought I recalled a while ago the bom.ini file was in the KiBom Master folder where that was the bom.ini it would generate new BOMs from... but now I can't find this file in the master folder it just generates one in the local project folder that you have to re copy and paste the configuration you want into each time which isn't efficient.

Let me know if I simply effed something up or if there is a better way to adjust or get around this.

Thanks

set-soft, 2022-11-02:
> Is KiBot another BOM plugin for KiCad?

KiBot is an automation script, you can use it to run KiBoM. It also has its own BoM generator, and can do a lot of other stuff.
The main target is Linux and is more oriented to be used in CI/CD environment, i.e. generate all the documentation after each GitHub commit.
To run it on Windows you need to use WSL or docker.
Here is an [example](https://inti-cmnb.github.io/kibot_variants_arduprog_site/Browse/t1-navigate.html) of what can be generated automatically, and here an [example](https://inti-cmnb.github.io/kibot_variants_arduprog/) of can you use it to implement variants (a feature missing in KiCad).
If you need to just generate a BoM then KiBot is an overkill ;-)

ch992319, 2022-11-02:
@set-soft 
Ahh ok I see, thank you for your help and maintaining KiBOM, extremely useful and easily the best BOM script I have used on KiCad. Appreciate you!

SchrodingersGat, 2022-11-07:
@ch992319 where do you input this extra "description" field? Normally the description comes from the symbol library and cannot be changed:

[image]

ch992319, 2022-11-07:
So typically I would go into here, which is in the home KiCad page in the dropdown menu "Preferences" and then select preferences in there:
[image]

Clicking preferences brings up this (more of a global allocation of symbol fields - so it automatically adds these fields to symbols for any symbol made in a schematic
[image]

Showing this in a test project after placing a random symbol down:
[image]

But with respect to your example, you could also hit the + button and add a symbol field and name it whatever you like:
[image]

Which when clicking does this:
[image]

Of which you may name whatever you like. Of course the first way shown gives it a consistent application across all projects.

However, now after trying this again in a new project to show what I mean't it looks like the problem has returned... I will show here - as you can see I have a "Description" field and "Descriptions" field, both say "HELLO WORLD"
[image]

But when I generate the bom (firstly here is my bom.ini file)
[image]

And now what shows up in the spreadsheet generated by the BOM script
[image]

Where you can see that Description is replaced by the library symbol description instead of the "Description" field I defined in the schematic symbol. Therefore my solution was to create another field called "Descriptions" and then add "Description" to the ignored fields in the bom.ini file, and add "Descriptions" to the column ordering of the bom.ini file.

Here is another BOM for an actual project that was configured this way to reach this end - HOW I WANT IT TO LOOK EXCEPT USING ONLY "DESCRIPTION":
[image]

However, @set-soft said that it will use the Description of the schematic symbol before the library one if there is something there to use... however after trying again (which I previously said did work) it now is not, and I think I mistakenly saw the "Descriptions" column but didn't extend it out all the way to see the "s" and thought it was the "Description" column. Which is why I thought it worked after @set-soft replied previously.

[image]
and
[image]

So what I am trying to say is that I want a column on my symbol called "Description" without the "s", and I want it to fill in that column with what I put in that field for the symbol in my schematic, not the description of that global library symbol...

ch992319, 2022-11-07:
Main thing is that I don't see what the benefit is to have the library description over the description field that you fill out... because I often use the same symbol for different components which use the same symbol but may require a different description - if that makes sense.

SchrodingersGat, 2022-11-07:
If you are expecting to be able to "override" the "Description" field here, I think that is an edge-case / undefined behaviour. The default KiCad field names should not be used as unexpected behaviour (like this) might happen. I would simply suggest that you keep a separate column with a different name

set-soft, 2022-11-07:
Hi @SchrodingersGat !

> I think KiBoM should use the lib description only when the component doesn't contain its own description.
> This is how [KiBot](https://github.com/INTI-CMNB/KiBot/) implements it.

I really think KiBoM should allow it. If the user defines `Description` with some value (not empty or just spaces) KiBoM should use the user value, otherwise use the value from the lib.
This issue seems to be recurrent. I think the reason is that people can't see the `Description` field in the component, and hence they define it.

ch992319, 2022-11-07:
> If you are expecting to be able to "override" the "Description" field here, I think that is an edge-case / undefined behaviour. The default KiCad field names should not be used as unexpected [...]

Is this a change however that I could petition for? haha! Perhaps for next update if and when one occurs? If not no worries I understand you're not getting paid to make this awesome plugin for the community - and I am extremely grateful for it in the first place, what exists already functionally exceeds all other plugins I have used.

set-soft, 2022-11-08:
Hi @SchrodingersGat and @ch992319 !

The following patch does the trick:

```diff
diff --git a/kibom/component.py b/kibom/component.py
index 577da0a..77e0e9e 100755
--- a/kibom/component.py
+++ b/kibom/component.py
@@ -184,6 +184,10 @@ class Component():
         return self.element.get("sheetpath", "names")
 
     def getDescription(self):
+        ret = self.element.get("field", "name", "description")
+        if ret:
+            return ret
+
         try:
             ret = self.element.get("libsource", "description")
         except:
```

With this patch the priority becomes:

1. User field named `Description`
2. From library found in the component (`libsource`)
3. From library (`libpart`)

And I think it makes sense.

SchrodingersGat, 2022-11-09:
See https://github.com/SchrodingersGat/KiBoM/pull/177

closed 2022-11-09 by SchrodingersGat