SchrodingersGat/KiBoM #178: LED footprint excluded from OBM

victorstoica114, 2023-02-14:
in KiCad libraries exists one footprint named "LED_SMD:LED_1206_3216Metric_ReverseMount_Hole1.8x2.4mm"
the problem is that this footprint is excluded from the BOM, most likely because it contains the syntax Mount_Hole
thanks!

SchrodingersGat, 2024-02-02:
You can adjust the regex to exclude these footprint as desired :)

closed 2024-02-02 by SchrodingersGat

victorstoica114, 2024-02-02:
I know, I've already done that. I opened this issue because someone who uses kibom for the first time and doesn't check the bom, will exclude that component from the order. that is, of course, in case he uses that footprint for the led. in any case, it's good that the developers know about the problem with that footprint. just in case...