SchrodingersGat/KiBoM #187: Error, mis-combining components

drmcnelson, 2023-05-17:
The following two items from the schematic are 30k resistots, but the first is  0.1%, 1/5W,  the second is 1% 1/4W.

[image]

[image]

KiBoM  combines them into one spreadsheet row,  and puts both partnumbers in the digikey column. 

[image]

Please advise.

drmcnelson, 2023-05-18:
Ok,  so I found the bom.ini and added a field "Description".

Now, the issue is a default location for the bom.ini,   perhaps .bom.ini in the login directory

SchrodingersGat, 2023-05-18:
Yep that a [.ini configuration option](https://github.com/SchrodingersGat/KiBoM#custom-column-grouping)

> Now, the issue is a default location for the bom.ini, perhaps .bom.ini in the login directory

? A default .ini file should be created already?

drmcnelson, 2023-05-18:
I found the default .ini in the project directory.    That is not so convenient.   I think what you want is the following (in schematic)

 if --cfg specified
   use it,
else if in project directory
  use it,
else if .kibom.ini in login directory or under .config/kibom/kibom.ini   
  use it
else 
   create it in local project directory and use it

drmcnelson, 2023-05-19:
BTW,  the program works well.   Very helpful.