SchrodingersGat/KiBoM #188: Feature Request: Set new DNP field based on configured variant

myamigo, 2023-05-29:
Thank for this script. It's very useful!

I'm wondering if the following could work as a way to better leverage support for the new DNP field in symbols. 

Rather than including the new DNP field as an input to your script as it does now with the new code added to **isFitted(()** could you instead set that field based on the DNP status as determined by the other tests in the same function (under # Variants logic). With this change when **isFitted()** exits the component's DNP flag will be explicitly set or cleared based on the variant field settings +BLAH, -BLAH.

After running the script for a particular variant the schematic would then visually display the DNP status for every component in the design. 

I tried to figure out where and how to hack some changes in to test this but I got stuck not knowing how to set a field attribute. 

Thank again!

set-soft, 2023-05-30:
Hi @myamigo !
Take a look at [KiBot](https://github.com/INTI-CMNB/KiBot), it can generate printed outputs of each variant, you can even generate new schematics and PCBs with the variants applied.
KiBot can use KiBoM to generate the BoMs, as a quick start option, but also implements a BoM output that is strongly inspired (read: I took a lot of @SchrodingersGat good ideas and code) that goes even farther. It supports KiBoM style variants, but also KiCost and iBoM variants.
It runs on Linux, WSL and docker images. Not plain vanilla Windows (i.e. without enabling WSL or without installing docker).

set-soft, 2023-05-30:
Hi @SchrodingersGat !

> with a way of implementing this seamlessly, let me know!
Lamentably the Python schematic API (in the roadmap for 6.0) was never implemented, so you must deal with the schematic format, which isn't well documented. KiBot can read and write schematic for KiCad 5, 6 and 7 (even for 7.99). So it can load a schematic, apply the variants to the DNP attribute and store it to disk. Is really complex.