SchrodingersGat/KiBoM #192: Part is excluded if footprint name is too long

KittyBumpkins, 2023-09-28:
Part is excluded from BOM if footprint name is too long.  Should default to include part in BOM.

SchrodingersGat, 2023-10-10:
@KittyBumpkins can you provide an example of a long footprint name which would be excluded?

KittyBumpkins, 2023-10-10:
Here is the footprint name that caused the problem:

DB9_Female_RA_EdgePinOffset7.70_MountingHolesOffset9.12

After the footprint name was shortened the part was included in the BOM.

SchrodingersGat, 2023-10-10:
```
python "C:\kicad\KiBoM/KiBOM_CLI.py" "C:\...\test_pro.xml" "C:/.../test_pro/test_pro" -vvvvv
Success.
 KiBOM version 1.9.0
 Output directory: 'C:\...\test\test_pro'
 PCB variant: default
INFO b"Excluding 'R1': Field 'Footprint' (DB9_Female_RA_EdgePinOffset7.70_MountingHolesOffset9.1) matched 'mount.*hole'"
INFO Removing: Build Quantity
```

It looks like it is simply matching the regex too aggressively, not related to a long footprint name

closed 2024-02-02 by SchrodingersGat