SchrodingersGat/KiBoM #193: Include Multiple Manufacturers and Part Numbers

dbeckwith01, 2023-11-01:
What's the best way to include alternate manufacturers and part numbers for a part and have KiBoM place this is the BOM?  For example I have MFG and MPN as fields on all parts but for some I want to add MFG1 and MPN1 in addition to MFG and MPN.  I looked at using the REGEX but that seems more for including or excluding particular parts.

Thanks,

SchrodingersGat, 2023-11-01:
You can always just add this information as extra fields?

dbeckwith01, 2023-11-10:
Ok thank you for that.  I assume I have to add those extra field names to the `[COLUMN_ORDER]` in `bom.ini` so they are written out?

SchrodingersGat, 2023-11-13:
They should be written out by default, `COLUMN_ORDER` is only if you want to rearrange them

closed 2024-02-02 by SchrodingersGat