SchrodingersGat/KiBoM #194: "Exclude from Bill of Materials" Still Included In the BOM

dbeckwith01, 2023-11-10:
I have several parts, SMD jumper pads and a Tag connect connector footprint marked with "Exclude from bill of materials" checked in the Symbol Properties but they still appear in KiBoM output.  Is there anyway to have KiBoM respect this flag?

SchrodingersGat, 2023-11-15:
@dbeckwith01 are you using the latest version? I have just checked with a sample schematic, and the component is not included in the generated BOM when `Exclude from bill of materials` is checked

dbeckwith01, 2023-11-18:
Hi @SchrodingersGat I'm using KiBot v1.6.3 in Docker via `docker pull setsoft/kicad_auto:ki7`

SchrodingersGat, 2023-11-19:
@dbeckwith01 can you confirm that the version `1.9.0` is being run, and that there's not some other version of the library installed?

dbeckwith01, 2023-11-22:
@set-soft 

Thank you for the help.  I'm new to using KiBot and KiBoM to produce my fabrication outputs.  Would you like me to open an issue in KiBoT with these details and include my KiBoT configuration?

set-soft, 2023-11-22:
Hi @dbeckwith01 !

Yes, and include a small example, just a few components where we can expose the issue.

SchrodingersGat, 2024-02-02:
Closed by #197

closed 2024-02-02 by SchrodingersGat